xxxajk/PhysicaloidLibrary #2: Support for CH340G Serial to USB Chips

xxxajk, 2018-08-01:
Bring forward from [this](https://github.com/ksksue/PhysicaloidLibrary/issues/30) issue.

DRSDavidSoft, 2018-08-01:
📌 **/CC** https://github.com/ksksue/PhysicaloidLibrary/issues/30 and https://github.com/xxxajk/PhysicaloidLibrary/pull/1#issuecomment-363651988

#### Original Comment:
> Some Arduino Clones use CH340G Serial to USB Chips, with DTR pin also connected.
> Could you take a look at the following drivers and add it to Physicaloid? That way we can upload HEX files using both CP chips and CH34x chips.
> [/usbserial/driver/Ch34xSerialDriver.java](https://github.com/mik3y/usb-serial-for-android/blob/master/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/Ch34xSerialDriver.java)
> [/usbserial/CH34xSerialDevice.java](https://github.com/felHR85/UsbSerial/blob/master/usbserial/src/main/java/com/felhr/usbserial/CH34xSerialDevice.java)

xxxajk, 2018-08-01:
Yes, the end goal is to be identical to the original API for other chips, therefore that will work after I am done.

xxxajk, 2018-08-02:
Just pushed untested code. No idea if it will work.
Please note that I deprecated "Readlisner" as it is a misspelling.
Use ReadListener instead to avoid compiler warnings.
You can ignore these warnings while compiling the library, though.

xxxajk, 2018-08-03:
Latest push is tested as "works for me".
Please give it a shot and let me know if it works for you.

xxxajk, 2018-08-03:
Please note that uploading won't work yet, but echoing does on a bare demo board. Probably still need to tickle some registers in the chip.

xxxajk, 2018-08-03:
[image]

DRSDavidSoft, 2018-08-03:
@xxxajk Can you provide pre-built .apk packages?

xxxajk, 2018-08-03:
@DRSDavidSoft 
It is actually easy to build.
Just include it as a library dependency build in your project, and the SDK will handle the rest.
I know it works with the old SDK (I'm using 32bit tools here and Netbeans) and should work just fine with the gradle junk supplied with the newer bloat from google.

xxxajk, 2018-08-03:
[image]

DRSDavidSoft, 2018-08-05:
@xxxajk Amazing, the CH340G chip with Physicalloid is working perfectly!
Thanks for your efforts, the only thing which is missing now is the ability to upload `.HEX` files.

On another note, I added Travis CI to the project (#4) which can build and upload .apk files automatically, since using the NetBeans IDE's builds after each commit could become a bit slow.

As as sidenote, do you use Ubuntu to build the project? The screenshot you added above doesn't seem to be Windows.

DRSDavidSoft, 2018-08-05:
I also updated the README via the same PR. Once again, thanks for making this possible :)

xxxajk, 2018-08-05:
Use PhysicaloidTest, or write your own code to upload. It isn't difficult to do... Thanks for the pull request, I'll merge it.

DRSDavidSoft, 2018-08-05:
Thanks!

One more question; which source code did you use as the driver for the project? Or did you write it from the Linux source yourself?

xxxajk, 2018-08-05:
Directly from Linux C sources, which I can do, since I am one of the orioginal Linux Kernel authors from the 1991 era. Linux 0.99 FTW! LOL. Yes, really. I've not been active in quite some time, but my name does live on in the source code, and documentation areas.

DRSDavidSoft, 2018-08-05:
@xxxajk Wow, That's so cool! It's like a superpower being able to port C code into Java so perfectly. I can only imagine, it must have been glorious days working on the early Linux, surely you must have worked with Linus Torvalds himself, at some time!

Although it makes me want to ask more questions! e.g. [ch341.c](https://github.com/torvalds/linux/blob/master/drivers/usb/serial/ch341.c) itself has problems in certain areas (it seems the DTR pin does not work correctly with this driver on Raspbian). I wonder how is it that it works on Android? What did you change during porting it?

xxxajk, 2018-08-06:
Probably an old kernel. It has been fixed long ago in the mainline kernel.

DRSDavidSoft, 2018-08-28:
@xxxajk I think this issue can now be marked as 'Solved', as your solution is working perfectly.
Tthanks for all the great help! I'll create a new one if I encounter any more issues.

xxxajk, 2018-09-28:
Ok, closing the issue, thanks for responding.

closed 2018-09-28 by xxxajk