xxxajk/PhysicaloidLibrary #3: Romove need for closed source FTDI jar file

xxxajk, 2018-08-01:
I have the knowledge on how to do this, and I plan on not having to depend on FTDI's cruft.
In short, I will be removing the need for the D2XX.jar file.

DRSDavidSoft, 2018-08-05:
I think I now know how you plan on doing this, using [ftdi_sio.c](https://github.com/torvalds/linux/blob/master/drivers/usb/serial/ftdi_sio.c), right? 😄 
Go for it, applaud for removing closed-source libraries!

xxxajk, 2018-08-06:
Partially, I also work on this: https://github.com/felis/UHS30 which is going to get WCH support soon.

xxxajk, 2018-09-28:
@DRSDavidSoft Will be working on this today. Will you be able to test?

DRSDavidSoft, 2018-09-29:
@xxxajk Yes, gladly! I need to find my FTDI adapter first.

xxxajk, 2018-09-29:
@DRSDavidSoft 
Initial testing shows 'works for me'. Give it a try. 

I'll be filling in the remaining missing features for CDC-ACM later today as well, namely the data size, parity and stop bits. These were never implemented even in the original library, and I know how to do it.

DRSDavidSoft, 2018-10-03:
@xxxajk Sorry for the delay, IRL stuff. I'll run my own tests today and update with you soon.