xxxajk/PhysicaloidLibrary #7: Cannot connect to device

fire1, 2018-08-21:
From several days i'm trying to connect to several serial devices but without a result.
I Added PhysicaloidTest to my project and this is the output from debug:
`D/UsbCdcConnection: Cannot find VID:1027, PID:0
                    Cannot find VID:3368, PID:0
                    Cannot find VID:3368, PID:0
                    Cannot find VID:4951, PID:0
                    Cannot find VID:5538, PID:0
                    Cannot find VID:4292, PID:0`

Deppendency lib are added: physicaloidlibrary.jar && d2xx.jar

xxxajk, 2018-08-21:
1: What devices are you attempting to connect exactly?
2: Are you using Linux or Windows for development?

fire1, 2018-08-21:
On widows 10 is compiled...

Mega with CH**  and other with ftdi clone
Also Uno with ftdi clone and original ftdi UART to USB 
None of them works

xxxajk, 2018-08-21:
Have you tried a different phone or tablet? If not, try, and see if that will work.
Some phones don't allow host mode, and some phones and tablets can have bugs. You may just be one of the unlucky people to have such a device.

fire1, 2018-08-22:
Adding library folder  to the project fix this issue!

closed 2018-08-22 by fire1