xxxajk/PhysicaloidLibrary #8: Broken serial read

fire1, 2018-08-28:
I'm struggling to get serial read to work... only output I'm having is:
>  Stream: ����������������������� 
Changing baud rate with method "setBaudrate()" did not help  (same baud rate for arduino and device 115200 && 9600 tested) 
also 
> new String(readBuffer, "UTF-8");  
did not fix it 

Is there some other way to fix it or debug it ?

fire1, 2018-08-29:
Changing baud rate in 

> PhysicaloidLibrary\src\com\physicaloid\lib\usb\driver\uart\UartConfig.java
Fixes the communication 
Changing from `setConfig` or `setBaudrate` didn't change it ...

hamidsani, 2018-09-03:
Make sure to call the setBaudrate method AFTER port is opened. If you look closer at setBaudrate method you'll find that when port is null (or it doesn't exist) then it just returns false.

xxxajk, 2018-09-03:
Correct. All failed calls return false. You should always check the return values.