xxxajk/PhysicaloidLibrary #9: Support for uploading via ArduinoISP [feature request]

tfrederick7, 2018-09-03:
I have been able to use the current board definitions, but I find myself wanting to upload via an external programmer such as Arduino as ISP.  I can put some time into it for development, but I have not found the right combination of information to do it.

xxxajk, 2018-09-03:
Should be fairly easy to implement.
I assume that you are wanting this to burn boot loaders or 32U4 or an ATtiny or something like that?
Let me know so I can duplicate what you are attempting to do.

tfrederick7, 2018-09-03:
Thank you for the quick response.   I am trying to bypass the bootloader for program size and startup speed purposes.  Specifically I am targeting an ATMega328P running at 3.3V and 8MHz.  I am using an Arduino Pro Mini from Sparkfun as the ISP, but any would do.   

Programming an UNO through another as an ISP should be an equivalent duplication.

xxxajk, 2018-09-04:
When I get a bit of free time, I can implement this easily.

tfrederick7, 2018-09-05:
Thank you.  If there is anything that I can do to test out the implementation.  Please let me know.