xxxajk/PhysicaloidLibrary #15: Uno with CH340 chip don't upload

ramonbss, 2018-11-30:
Write/Read functions work fine with this chip, but every time I try to upload, the library returns that it couldn't initialize MCU.

ramonbss, 2018-12-08:
Tried everything I could to figure out the problem. Stk500 implementation worked fine in all arduino models I've tested. And the problem not seems related to CH34x driver since its Write/Read flawlessly with the board. In the Thread [ Support for CH340G Serial to USB Chips #2 ](https://github.com/xxxajk/PhysicaloidLibrary/issues/2) is said that maybe some registers have to be tickle. Hope it's a simple thing to implement.

xxxajk, 2018-12-09:
Last time I tested this was with the mega2560 containing the ch340g, and there have been no changes to it since. 
Don't forget you need to choose the correct MCU.
If you can show me the avrdude output, and it says there is a problem, then your board is bad.

ramonbss, 2018-12-09:
The board is fine and MCU is correctly selected. I mean that a tried to upload from IDE and read its output to compare with your java code but could not understand the code. My Uno CH34x works fine using Arduino IDE 1.8.7 but it unfortunately even start to upload using Physicaloid. I even tried using Physicloid Test but it didn't work either.
The library connects with the board, read its vendor and product id, recognize it as CH34x but fail to start uploading.

I'm developing a programming IDE based in blocks to help people in my community enter in this electronic/programming world. Since boards with ch34x are cheaper, it's crucial for the app to be compatible with them.

From the far that I could read/understand the library, I could see that you made an amazing job there. Everything very organized and working flawlessly!!

A screenshot from my Android's app ( I put your name in Playstore app detail as the developer of the library being used )

[image]
[image]

ramonbss, 2019-04-05:
Everything is fine!!!

closed 2019-04-05 by ramonbss

xxxajk, 2019-04-05:
Great to hear! :-)

xxxajk, 2019-04-05:
Just took a look on Play Store, says it is incompatible with all of my devices. I think you need to fix your manifest :-)