xxxajk/PhysicaloidLibrary #19: Ability to upload a firmware on STM32 Arduino board

bdureau, 2019-12-28:
I am looking at a way to upload a firmware on an stm32f103c8t6 board that has an Arduino bootloader  (Roger Clark https://github.com/rogerclarkmelbourne/Arduino_STM32)
This should in theory use the CdcACM driver because it is recognized as a Mapple USB on my PC.
Also I have noticed some issues when sending data to the board, I can send them (I have some code in the firmware that uses them) but then it is stuck in the mConnection.requestWait().
When I get that fixed I can have a look at the mapple loader program to use instead of the AVR
Can anybody help?
This should be marked as an enhancement request

xxxajk, 2019-12-28:
You will need to implement the protocol used to do this.
I don't have the maple, but I do have a discovery board here someplace. Never use it.

bdureau, 2022-01-25:
For those interested I have implemented it in the following app
https://github.com/bdureau/BearConsole2
This app contains a port of the python program that can flash the stm32

closed 2022-05-30 by xxxajk