xxxajk/PhysicaloidLibrary #22: Really hoping someone can help, even thought it seems like this lib is dead

pablo67340, 2021-04-17:
Hey there,

I am trying to flash my Arduino Nano from my Android. I tried Hex with bootloader, and without bootloader, both same results.

Anyways, here is my avrdude specs for my arduino.

`Using Port                    : COM11
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

`

I have my board set to Boards.ARDUINO_NANO_328 as this is what I have the board set to in my Arduino IDE, which programs the board fine.

For some reason, this is the response I am getting from trying to upload the hex.

`2021-04-17 02:28:38.481 25137-12127/com.example.user.navigationdrawersample D/UartWinCH34x: Opening WCH
2021-04-17 02:28:38.483 25137-12127/com.example.user.navigationdrawersample D/Physicaloid: upload : open successful
2021-04-17 02:28:38.483 25137-12127/com.example.user.navigationdrawersample E/MapWithMarker: Upload : Start
2021-04-17 02:28:38.483 25137-12128/com.example.user.navigationdrawersample D/UsbRequestJNI: init
e[512689868192]
2021-04-17 02:28:41.850 25137-12127/com.example.user.navigationdrawersample E/Stk500: recv timeout.
2021-04-17 02:28:41.850 25137-12127/com.example.user.navigationdrawersample E/Stk500: STK500.getsync(): not in sync: resp=0x00
2021-04-17 02:28:43.102 25137-12127/com.example.user.navigationdrawersample E/Stk500: recv timeout.
2021-04-17 02:28:43.102 25137-12127/com.example.user.navigationdrawersample E/Stk500: STK500.getparm(): (a) protocol error, expect=0x14, resp=0x41
2021-04-17 02:28:43.353 25137-12127/com.example.user.navigationdrawersample E/Stk500: recv timeout.
2021-04-17 02:28:43.353 25137-12127/com.example.user.navigationdrawersample E/Stk500: STK500.getparm(): (a) protocol error, expect=0x14, resp=0x41
2021-04-17 02:28:43.604 25137-12127/com.example.user.navigationdrawersample E/Stk500: recv timeout.
2021-04-17 02:28:43.604 25137-12127/com.example.user.navigationdrawersample E/Stk500: STK500.initialize(): (a) protocol error, expect=0x14, resp=0x42
2021-04-17 02:28:43.604 25137-12127/com.example.user.navigationdrawersample E/AvrUploader: initialization failed (-1)
2021-04-17 02:28:43.604 25137-12127/com.example.user.navigationdrawersample E/MapWithMarker: Error  : Can't initialize MCU.
2021-04-17 02:28:43.604 25137-12127/com.example.user.navigationdrawersample E/MapWithMarker: Upload fail
2021-04-17 02:28:43.606 25137-12127/com.example.user.navigationdrawersample D/UsbAccessor: Close(0)
2021-04-17 02:28:43.606 25137-12127/com.example.user.navigationdrawersample D/UsbDeviceConnectionJNI: close`

I would really appreciate any support with this. I get the feeling the odds are I wont get any replies, but I can hope!

xxxajk, 2021-04-17:
Far from dead, it's currently used in real-world products.
Now on to the question :-)
I've not a nano, so I can't actually test on one, but the messages are saying that it's not successful talking to the bootloader.
Usually this means the baud rate is incorrect, or you need to use STK500-v2.

pablo67340, 2021-04-17:
This seems like the cause, however, Boards has Private Access, so I am unable to instantiate a custom Board with custom values, such as the UploadProtocol.

xxxajk, 2021-04-17:
:-) You don't have to define a custom board... but you are welcome to do so if you want. 
Are you building from source? Pull requests are always welcome here.

pablo67340, 2021-04-28:
I needed to declare a custom board as I couldn't find the method to change the upload type. It was easier to declare a custom board, so I can provide all custom values at once. I am not sure if my nano baud rate is different than what you have in the enum because it's made by a different company? Not entirely sure if it would be worth a PR if it's only going to mess up people who have real arduino nano's with the proper baud rate.

xxxajk, 2021-04-28:
Won't be a problem if you just add your custom board to the list here:
https://github.com/xxxajk/PhysicaloidLibrary/blob/master/PhysicaloidLibrary/src/com/physicaloid/lib/Boards.java#L19-L58

Is this going to be a product on the market or something?