xxxajk/PhysicaloidLibrary #23: Arduino Nano ATmega328 (old bootloader) cannot open connection

davidezanella, 2021-08-19:
It seems that the library is not able to open a connection with my Arduino Nano ATmega328 (old bootloader) even if it is correctly identified. It is something related to the UART protocol, I suppose. The following is the avrdude output while uploading from Arduino IDE:
```
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM9
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 2
         Firmware Version: 1.16
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

```

This is the log of the library while trying to open a connection: (I changed some log to report more data)
```
D/AutoCommunicator: POSSIBLY WCH
    Yes WCH!
D/UartWinCH34x: Trying VID: 6790
D/UsbCdcConnection: VID PASS 6790
    PID PASS 29987
    ACM PASS
    ch == chNum
    deviceIsConnected 0
D/UsbAccessor: Device(0) : UsbDevice[mName=/dev/bus/usb/001/002,mVendorId=6790,mProductId=29987,mClass=255,mSubclass=0,mProtocol=0,mManufacturerName=null,mProductName=USB2.0-Ser!,mVersion=2.54,mSerialNumberReader=android.hardware.usb.IUsbSerialReader$Stub$Proxy@505625,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=48,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=255,mSubclass=1,mProtocol=2,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=32,mInterval=0]
    UsbEndpoint[mAddress=2,mAttributes=2,mMaxPacketSize=32,mInterval=0]
    UsbEndpoint[mAddress=129,mAttributes=3,mMaxPacketSize=8,mInterval=1]]]]
D/UsbAccessor: Device(0) : UsbDevice[mName=/dev/bus/usb/001/002,mVendorId=6790,mProductId=29987,mClass=255,mSubclass=0,mProtocol=0,mManufacturerName=null,mProductName=USB2.0-Ser!,mVersion=2.54,mSerialNumberReader=android.hardware.usb.IUsbSerialReader$Stub$Proxy@60badab,mConfigurations=[
[... 48 lines cut ...]
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=32,mInterval=0]
    UsbEndpoint[mAddress=2,mAttributes=2,mMaxPacketSize=32,mInterval=0]
    UsbEndpoint[mAddress=129,mAttributes=3,mMaxPacketSize=8,mInterval=1]]
    Endpoint(0,0,0) : UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=32,mInterval=0]
D/UsbAccessor: Device(0) : UsbDevice[mName=/dev/bus/usb/001/002,mVendorId=6790,mProductId=29987,mClass=255,mSubclass=0,mProtocol=0,mManufacturerName=null,mProductName=USB2.0-Ser!,mVersion=2.54,mSerialNumberReader=android.hardware.usb.IUsbSerialReader$Stub$Proxy@8d610b4,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=48,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=255,mSubclass=1,mProtocol=2,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=32,mInterval=0]
    UsbEndpoint[mAddress=2,mAttributes=2,mMaxPacketSize=32,mInterval=0]
    UsbEndpoint[mAddress=129,mAttributes=3,mMaxPacketSize=8,mInterval=1]]]]
D/UsbAccessor: Interface(0,0) : UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=255,mSubclass=1,mProtocol=2,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=32,mInterval=0]
    UsbEndpoint[mAddress=2,mAttributes=2,mMaxPacketSize=32,mInterval=0]
    UsbEndpoint[mAddress=129,mAttributes=3,mMaxPacketSize=8,mInterval=1]]
    Endpoint(0,0,1) : UsbEndpoint[mAddress=2,mAttributes=2,mMaxPacketSize=32,mInterval=0]
D/UartWinCH34x: Opening WCH, VID: 6790
    init
D/UartWinCH34x: init r != 2 for 0x5f, r: -1 size: 8 buff: [0, 0, 0, 0, 0, 0, 0, 0]
    Trying VID: 17224
D/UsbCdcConnection: Cannot find VID:17224, PID:0
D/UartWinCH34x: Opening WCH failed
D/AutoCommunicator: Nothing on USB
```

What I get is that it doesn't receive any data from the Arduino during the init phase. Can someone please help me understanding the problem or redirecting me to some documentations/specifications?

xxxajk, 2021-08-19:
` D/UartWinCH34x: init r != 2 for 0x5f, r: -1 size: 8 buff: [0, 0, 0, 0, 0, 0, 0, 0]
`
seems to be the bug with your hardware...

https://github.com/xxxajk/PhysicaloidLibrary/blob/master/PhysicaloidLibrary/src/com/physicaloid/lib/usb/driver/uart/UartWinCH34x.java#L263

:-)

xxxajk, 2021-08-19:
Actually, what is the VID:PID?

davidezanella, 2021-08-19:
The VID is: 1A86 and PID: 7523

xxxajk, 2021-08-19:
Found the issue, I'll have time on Sunday to fix it. Thanks for locating the bug!

jangminhyeok, 2021-12-07:
@xxxajk I strongly believe that it is because of ch341_control_in & ch341_control_out ( UsbDeviceConnection.controlTransfer() ).
mconnection was not null, but r (ret) returns -1.

#1 : https://github.com/mik3y/usb-serial-for-android/blob/master/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/Ch34xSerialDriver.java
#2 : https://github.com/felHR85/UsbSerial/blob/master/usbserial/src/main/java/com/felhr/usbserial/CH34xSerialDevice.java

those libraries are usb serial for android, and they prepared several initiation progress with various hexadecimal code in case of failure. I don't know much about UART communication, so I cannot fix the code and contribute.

`12-07 15:45:19.341 24598 24784 I ReactNativeJS: Running "main" with {"rootTag":1}
12-07 15:45:41.227 24598 24784 I ReactNativeJS: 'returnSketch State: ', 'uploadSketch'
12-07 15:45:41.228 24598 24784 I ReactNativeJS: 'Generated Code: \n', '// generated by K-STEAM Coding\nvoid setup() {\n}\n\nvoid loop() {\n}\n'
12-07 15:45:41.532 24598 24784 I ReactNativeJS: 'Permission Status: ', { granted: true,
12-07 15:45:41.532 24598 24784 I ReactNativeJS:   canAskAgain: true,
12-07 15:45:41.532 24598 24784 I ReactNativeJS:   status: 'granted',
12-07 15:45:41.532 24598 24784 I ReactNativeJS:   expires: 'never' }
12-07 15:45:41.543 24598 24784 I ReactNativeJS: 'File Uri: ', 'file:///data/user/0/com.ksteam.blockly/files/2141IQZnxYwsyEDZJeXKQb2L.ino'
12-07 15:45:41.543 24598 24784 I ReactNativeJS: 'Genearted Ino file (during _upload function debugger): ', 'file:///data/user/0/com.ksteam.blockly/files/2141IQZnxYwsyEDZJeXKQb2L.ino'
12-07 15:45:41.606 24598 24784 I ReactNativeJS: 'Uploaded progress: ', 100, '%'
12-07 15:45:42.124 24598 24784 I ReactNativeJS: 'Download progress: ', 100, '%'
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :100000000C9434000C9446000C9446000C9446006A
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :100010000C9446000C9446000C9446000C94460048
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :100020000C9446000C9446000C9446000C94460038
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :100030000C9446000C9446000C9446000C94460028
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :100040000C9448000C9446000C9446000C94460016
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :100050000C9446000C9446000C9446000C94460008
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :100060000C9446000C94460011241FBECFEFD8E03C
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :10007000DEBFCDBF21E0A0E0B1E001C01D92A930FC
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :10008000B207E1F70E9492000C94DC000C9400008F
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :100090001F920F920FB60F9211242F933F938F93BD
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :1000A0009F93AF93BF938091050190910601A0911A
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :1000B0000701B09108013091040123E0230F2D378F
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :1000C00058F50196A11DB11D2093040180930501EF
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :1000D00090930601A0930701B0930801809100015D
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :1000E00090910101A0910201B09103010196A11D1F
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :1000F000B11D8093000190930101A0930201B09380
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :100100000301BF91AF919F918F913F912F910F90DC
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :100110000FBE0F901F90189526E8230F0296A11D81
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :10012000B11DD2CF789484B5826084BD84B58160DE
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :1001300084BD85B5826085BD85B5816085BD8091B2
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :100140006E00816080936E0010928100809181002A
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :100150008260809381008091810081608093810022
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :10016000809180008160809380008091B1008460E4
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :100170008093B1008091B00081608093B000809145
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :100180007A00846080937A0080917A008260809304
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :100190007A0080917A00816080937A0080917A0061
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :1001A000806880937A001092C100C0E0D0E0209770
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :0C01B000F1F30E940000FBCFF894FFCF99
12-07 15:45:42.127 24598 24784 I ReactNativeJS: :00000001FF
12-07 15:45:42.253 24598 24784 I ReactNativeJS: 'Permission Status: ', { granted: true,
12-07 15:45:42.253 24598 24784 I ReactNativeJS:   canAskAgain: true,
12-07 15:45:42.253 24598 24784 I ReactNativeJS:   status: 'granted',
12-07 15:45:42.253 24598 24784 I ReactNativeJS:   expires: 'never' }
12-07 15:45:42.261 24598 24784 I ReactNativeJS: 'File Uri: ', 'file:///data/user/0/com.ksteam.blockly/files/2142dwYgIDPI5xh947v4Qc4n.hex'
12-07 15:45:42.261 24598 24784 I ReactNativeJS: 'Genearted Hex file (during _upload function debugger): ', 'file:///data/user/0/com.ksteam.blockly/files/2142dwYgIDPI5xh947v4Qc4n.hex'
12-07 15:45:42.262 24598 24784 I ReactNativeJS: 'Debugging after _createHex code: ', 'string'
12-07 15:45:42.262 24598 24784 I ReactNativeJS: /data/user/0/com.ksteam.blockly/files/2142dwYgIDPI5xh947v4Qc4n.hex
12-07 15:45:42.264 24598 24784 I ReactNativeJS: Uploading Done
12-07 15:45:42.266 24598 24785 D ReactNativePhysicaloid: FilePath: /data/user/0/com.ksteam.blockly/files/2142dwYgIDPI5xh947v4Qc4n.hex
[... cut ...]
12-07 15:45:57.592 24598 24785 D UartWinCH34x: Opening WCH failed
12-07 15:45:57.593 24598 24785 D UsbCdcConnection: Cannot find VID:17224, PID:0
12-07 15:45:57.593 24598 24785 D UartWinCH34x: Opening WCH failed
12-07 15:45:57.593 24598 24785 D AutoCommunicator: Nothing on USB
12-07 15:45:57.594 24598 24785 D ReactNativePhysicaloid: Uploading Completed
12-07 15:45:57.594 24598 24785 D ReactNativePhysicaloid: mPhyscaloid.upload function done
12-07 15:45:57.594 24598 25109 D Physicaloid: upload : mSerial is null`

jangminhyeok, 2021-12-08:
I fixed this problem with below codes in 'UartWinCH34x.java'.

`
// USB_TYPE_VENDOR | USB_RECIP_DEVICE | USB_DIR_OUT
private static final byte REQTYPE_HOST_TO_INTERFACE = (byte) (UsbConstants.USB_TYPE_VENDOR | UsbConstants.USB_DIR_OUT);
// USB_TYPE_VENDOR | USB_RECIP_DEVICE | USB_DIR_IN
private static final byte REQTYPE_INTERFACE_TO_HOST = (byte) (UsbConstants.USB_TYPE_VENDOR | UsbConstants.USB_DIR_IN);
`

xxxajk, 2021-12-09:
Please make a pull request.