xxxajk/PhysicaloidLibrary #24: Converted to React Native Module, but doesn't work properly

jangminhyeok, 2021-09-01:
I'm trying to turn Android library into a React Native module, but there's no error, but it's not working properly.

The above library provides the ability to upload Arduino firmware Hex code by Android phone (written in Java). So I compressed it into a JAR file and approached it in the following way.

Add jar files to android/app/libs and set gradle dependencies
Create and code RNPhysicaloidModule.java and
RNPhysicaloidPackage.java within android/app/src/main/java
Done adding to the getPackages() part of MainApplication.java
In App.tsx, added import { NativeModules } from 'react-native';
call function with "await NativeModules.RNPhysicaloid.flashHexFirmware(filepath);"
After running the app through this process, I connected the Arduino board to the OTG cable and verified that otg usb cable was properly connected through the react-native-serialport library.

However, I ran a function called flashHexFirmware by pressing certain buttons, but found that there was no response to the Arduino (same in STK500, CH340).

I'm afraid I've got the wrong approach to creating a native module, and I'm looking for advice...

code can be seen here : https://stackoverflow.com/questions/69009563/react-native-module-with-existed-android-library-but-doesnt-work-properly

jangminhyeok, 2021-09-01:
[image]

I find out that, It was because of can't opening the firmware file.
I think I can fix it, then I will contribute the react-native-physicaloid library later.

btw Special thanks to xxxajk!

xxxajk, 2021-09-03:
gradle... fun... NOT! :-) I use Apache ANT because it allows a lot of things gradle/studio doesn't, such as embedding binary data into the APK (e.g. pre-decoded hex files) which studio/gradle will refuse to do. That' basically the main reason I don't use either of them.

There are a few work arrounds. 

You can: 
1. Allow your app to access external storage via permissions, and then select from external storage. It's called "external" even if it's internal. Part of the issue where the file actually is located is depending on the Android version. Depending on if the device maker (phone/tablet/watch) follows the rules, depends on how you find out where internal and external storage actually is mounted. *normally* there's an environment variable set which gives the hint, depending on the version, but there's never been a clear universal way to find this out. Basically you try several paths, and try another on each fail.
2. If you have your app download and then store the hex or binary file in the APK's PRIVATE DATA area, then there's not an issue, and only network permissions are needed. Of course you need to write the code to do this, and it's not too difficult. The problem is that you must have network access. If you don't have it, you are kind of screwed, unless you work something out that can do this via Bluetooth.

CH34x needs a little bit of love. A fix is known, but right now I'm very overloaded with work, which, funny enough, actually uses this library in a commercial product. Not much attention has been given to it because my client's product uses FTDI parts. I'm trying to get some time to dedicate to fixing it, hopefully this Saturday, providing my client doesn't need me to ship more hardware. Yours is the second request in as many weeks for CH34X fixes. I know what the fix is, just need the few minutes to apply it, test it, and push it. This pushes it up in the queue another notch in my TO-DO list.