xxxajk/PhysicaloidLibrary #30: Bad communication when Serial.print() without delay() command.

rkqrkqdl, 2022-11-29:
Hello. I'm a Korean beginner developer who wants to implement a serial monitor on an Android device.
I'm not good at English. So, I'm writing with the help of Google Translator, so please understand even if the sentences are strange.

My serial monitor works fine in most cases.
For example, the text sent by the Arduino board using Serial.println() is well received through the ReadListener of Physicaloid.

However, when the Arduino board sends text to Serial.print() without delay(), my serial monitor receives the wrong text.
I entered the command Serial.print("A"); on the Arduino Uno board. Then, I confirmed that the A was printed correctly in the serial monitor of the Arduino IDE. (ex. AAAA...)
But, the A was not printed correctly on my serial monitor. So, I used Logcat to check what values ​​the buffer received.
[image]
The buffer first receives the byte value of 102, and then continues to receive the byte value of 5.

I think the buffer receives the binary value incorrectly.
A => 65 (in ASCII) => 01000001

(Arduino Uno Send) AAAAA... => 01000001 01000001 01000001 01000001 01000001 ...

(Arduino IDE Buffer receive correctly)  01000001 / 01000001 / 01000001 / 01000001 / 01000001 / ... => AAAAA...
(My Buffer receive incorrectly) 01/000001 01/000001 01/000001 01/000001 01/000001 01/... => ?????
                                                           (= 5)          (= 5)           (= 5)           (= 5)          (= 5)

Umm... Why is my buffer wrong? I just followed PhysicaloidLibrary's SampleProjects.

While I was thinking about this problem, I found one difference.
When the serial monitor of the Arduino IDE is opened, the Arduino board is restarted.
But when I open my serial monitor it doesn't.
Could this be the reason?
If the reason is correct, how to restart the arduino board when opening the serial monitor? If not the reason, what other reason?

I would appreciate if you could tell me how to do this.
Thank you very much for your time.

xxxajk, 2022-11-30:
See the examples.
You must toggle DTR low then high to perform an Arduino reset.
Also, make sure that the the serial protocol is set correctly, 8/N/1 (8 bit word, No parity, one stop bit)

rkqrkqdl, 2022-12-01:
Thank you for your information! :)

closed 2024-03-18 by rkqrkqdl