xxxajk/PhysicaloidLibrary #31: How to add the library

senchoi, 2023-02-05:
the instruction seems in need of update.
can you just provide jar file?

xxxajk, 2023-02-08:
To build with ant on Linux (Sorry, don't use anything else, pretty sure it is a requirement anyway)
You do not need to  build for ART or newer JDK, or use any of the android studio tools or the horrible gradle junk.
You will need ibm-java80-jdk, and apache-ant
```
cd ~/PATH_TO/PhysicaloidLibrary/PhysicaloidLibrary
JAVA_HOME=/usr/lib/jvm/ibm-java80-jdk-x86_64 ant debug
```

Let me know if that works out for you.

senchoi, 2023-02-08:
I got it to work on Android Studio eventually.
By updating deprecated commands to suggested ones and following https://youtu.be/DG7OYSTxPUA partially.